idea-fasoc/OpenFASOC #1: Setup CI that automatically publishes some example IP

mithro, 2021-03-25:
It would be good to setup a CI system which automatically publishes some IP to a repository as part of the skywater-pdk.

Once you have it all working, we can probably create a https://foss-eda-tools.googlesource.com/skywater-pdk/ip/sky130_fasoc_ip_tempsensor/ repository.

msaligane, 2022-11-14:
@saicharan0112 and @luccareinehr I think we need to discuss how to do this. We have everything now to this.

saicharan0112, 2022-11-14:
I don't find any reference IPs that are published under this URL (at least
not publicly available). Do you happen to have any such?

saicharan0112, 2023-03-10:
I don't think this is relevant anymore. I neither see any IPs under that link and nor open the link https://foss-eda-tools.googlesource.com/skywater-pdk/ip/. Maybe we can have any example on how other IPs are backed? Is this for having a IP database similar to opencores?

closed 2023-03-22 by saicharan0112