idea-fasoc/OpenFASOC #2: LDO Vin range is not useful for powering sky130A standard cells

shalan, 2021-05-12:
According to [this](https://github.com/idea-fasoc/OpenFASOC/blob/main/generators/ldo-gen/tools/supported_inputs.json), the LDO generator cannot generate 1.8V LDO needed to power digital macros implemented using sky130 SCL (e.g., hd, hell, ...)

msaligane, 2021-05-12:
That needs to be updated. FYI, the LDO has not been released yet using the open source tools.

msaligane, 2021-05-12:
@wenbodd can you update these:
```
      "vin": {
        "max": 1.3,
        "min": 0.6}
```

shalan, 2021-05-12:
Is it characterize for Vin beyond 1.3V? The FASoC [website](https://fasoc.engin.umich.edu/digital-ldo/) lists the same range (0.6 to 1.3V) on the LDO specs table.

msaligane, 2021-05-12:
Yes, the sky130 version of the LDO takes in 3.3v/5v and generated 1.8v. These values from other technologies we are supporting and it seems the values were not updated.

I would advise not using this code for now. I will update this when it is ready.

mithro, 2021-05-12:
Can you please add an example which generates something useful for SKY130?

msaligane, 2021-05-12:
Sure. 
The temp sensor is the only thing released for now and the example is shown in the readme. LDO and SAR ADC are wip.
We should have updates around early June.

msaligane, 2022-11-14:
This should be fixed now. @lulu9312 ?

lulu9312, 2023-02-17:
Yes @msaligane  This is fixed. We can close this issue.

closed 2023-03-10 by saicharan0112