idea-fasoc/OpenFASOC #12: jupyter notebook as an alternative to matlab for testsetup

proppy, 2021-12-16:
@mithro pointed me to #11 where matlab based tests seems to be a lot of calculation and plotting to validate the design, I'm curious if you've considered using https://jupyter.org/ as an alternative (maybe thru https://github.com/mwouts/jupytext to make it source control friendly).

Note sure how easy it would be to replication the various calculation in Python, but Jupyter also have support for Octave (GNU/Matlab) https://github.com/Calysto/octave_kernel and Julia https://github.com/JuliaLang/IJulia.jl

msaligane, 2021-12-17:
Thanks for your pointers. MATLAB codes were more for post-processing of our own research publication purposes, and they are actually not part of the test setup and also not documented, so, for people who want to re-use our setup I am not sure whether they would also want to use the same post-processing or would have different methodologies - we have followed the one agreed on in literature. 

Now, to follow the reproducibility conventions in the software community, I think it is definitely good to put the MATLAB codes there, just not sure whether people need the Python version of them.

On translating those to Python, I think it should not be difficult as the MATLAB codes are mainly about data loading, cleaning, array manipulation, sorting and Min/Max finding, which I believe can also be implemented in Numpy. Except there is a systematic error correction step that uses an optimization solver built in MATLAB, for which I believe an equivalent Python library can be found out there. But to be frank that is a lot of code and I am afraid that @QiruiZhang or me won't have time for that translation in a while. 

We would really appreciate using your help in doing that! 
@QiruiZhang please feel free to add any comments.

proppy, 2021-12-17:
@msaligane thanks for providing more background on the matlab code, understood that those are not mean to be directly used by downstream user of the testing flow, and were primarily added as a reference to reproduce/explain the result and visualization from the research paper.

do you think it would make sense to move them outside of the `testsetup` directory then? or they strongly tied to the test project described in the https://github.com/idea-fasoc/OpenFASOC/blob/main/tapeouts/MPWI/testsetup/README.md

additionally to take you on the offer of converting some of those to python, I definitely think it would be a nice addition to have a notebook that allow to reproduce the example "Temperature Sensor" flow end to end, should we re-purpose this issue to track this?

msaligane, 2021-12-17:
@proppy thanks for your interest!

> do you think it would make sense to move them outside of the testsetup directory then? or they strongly tied to the test project described in the https://github.com/idea-fasoc/OpenFASOC/blob/main/tapeouts/MPWI/testsetup/README.md

I am okay moving them somewhere else or even take them out. Do you can send a PR for that?

> additionally to take you on the offer of converting some of those to python, I definitely think it would be a nice addition to have a notebook that allow to reproduce the example "Temperature Sensor" flow end to end, should we re-purpose this issue to track this?

Absolutely! I think we have started a notebook somewhere (I can't find it just yet) but it did not have the testing steps in. How do you think we should re-purpose this issue?

saicharan0112, 2022-10-10:
#102 #99 addressed this issue in its final stages

closed 2022-10-10 by saicharan0112