idea-fasoc/OpenFASOC #38: OpenFASOC should expose the main generator flow as an importable python modules

proppy, 2022-03-08:
In order to be able to orchestrate generation from other python codebase (or jupyter notebooks) it would be nice if OpenFASOC exposed a python interface over the main generators flows that can be `import`ed, https://github.com/idea-fasoc/OpenFASOC/blob/main/generators/temp-sense-gen/tools/temp-sense-gen.py

Note:  https://github.com/idea-fasoc/OpenFASOC/blob/main/generators/temp-sense-gen/tools/TEMP_netlist.py seems to already exposes functions some lower level function that could be imported, but it seems that there is a fair amount of work also done from the outer scripts.

msaligane, 2022-03-08:
Thanks @proppy 
This is really great!
So let's say if a generator is updated, what would be the procedure to re-package it as a python module? is there an automated way?

proppy, 2022-03-08:
for conda packages that could happen as part of https://github.com/hdl/conda-eda/issues/192 (the project has a continuous integration environment that rebuild package from HEAD every night) or thru dedicated https://github.com/features/actions in this repo if we wanted to publish those package on PyPI too.

msaligane, 2022-03-09:
@erictaur has been working on github actions and somehow got it to work. @erictaur any updates?

mithro, 2022-03-09:
We could make this work now with https://github.blog/2021-11-29-github-actions-reusable-workflows-is-generally-available/ (rather than my older https://github.com/mithro/actions-includes).

mithro, 2022-03-09:
I believe these are deployed on https://github.com/chipsalliance/fasm/actions

joamatab, 2022-04-23:
See PR #43  for this

I can also add some jupyter notebooks for the documentation

msaligane, 2022-04-23:
That would be great! let's schedule some time early this week. Thanks Joaquin!

saicharan0112, 2022-09-08:
@proppy suggested a good idea to expose each stage inside the generator as a python module which can end up something like this
Eg:
```
from openfasoc import tempsense
netlist=tempsense.gen_netlist()
tempsense.run_synthesis(netllist...)
```
Next steps could be interesting after we have something. @proppy do you want to add something to this?

msaligane, 2022-09-08:
Sounds great to me!

saicharan0112, 2022-10-23:
Here is the idea:
1. Convert the three verilog netlists (TEMP_ANALOG_hv.v, TEMP_ANALOG_lv.v and counter_generic.v present in [folder](https://github.com/idea-fasoc/OpenFASOC/tree/main/openfasoc/generators/temp-sense-gen/src) ) into python modules to generate TEMP_ANALOG_hv_nl.v, TEMP_ANALOG_lv_nl.v and counter.v files.. I could just think of using print statements to build these files because of their size. Any suggestions?

2. Convert TEMP_netlist.py, which generates the tempsense netlist using the function gen_temp_netlist(), into importable python module.

In the end, I think we would be having something like this `from openfasoc import gen_temp_netlist`. We could add other netlists in the future. For eg: `from openfasoc import gen_ldo_netlist` or `from openfasoc import gen_pll_netlist`. 

Just having the netlist, I think we could represent the tempsense flow in notebook. 

Looks simple, does simple thing. Do you people think that this is worth?

proppy, 2022-10-24:
>  I could just think of using print statements to build these files because of their size. Any suggestions?

Curious if you're thinking of using an HDL for helping you maintain a structured view of the netlists, see: https://github.com/drom/awesome-hdl for some recommendations (https://github.com/amaranth-lang/amaranth, https://github.com/PyHDI/veriloggen comes to mind or https://github.com/PyHDI/Pyverilog and https://github.com/byuccl/spydrnet as lower leve option).

If you prefer to manipulate raw text, I would recommend to look at the templating language like https://jinja.palletsprojects.com/en/3.1.x/.

saicharan0112, 2022-10-26:
> 

Thanks.. Understood. I am not familiar with those projects who shared here though. I will take a look.