idea-fasoc/OpenFASOC #63: python dependencies installtion on centos7

saicharan0112, 2022-05-29:
@joamatab I am trying to install python dependencies using `make install` but I am getting this below error. Do you have any idea on this?
```
Collecting gdsfactory==5.1.1 (from -r requirements.txt (line 5))
  Could not find a version that satisfies the requirement gdsfactory==5.1.1 (from -r requirements.txt (line 5)) (from versions: 1.1.1, 1.1.3, 1.1.5, 1.1.6, 1.1.8, 1.1.9, 1.2.2, 1.3.2, 1.4.0, 1.4.2, 1.4.3, 2.0.0, 2.0.2, 2.1.0, 2.1.1, 2.1.2, 2.1.3, 2.1.4, 2.2.1, 2.2.2, 2.2.3, 2.2.4, 2.2.5, 2.2.6, 2.2.8, 2.2.9, 2.3.1, 2.3.2, 2.3.3, 2.3.4, 2.4.1, 2.4.2, 2.4.3, 2.4.4, 2.4.5, 2.4.6, 2.4.7, 2.4.8, 2.4.9)
No matching distribution found for gdsfactory==5.1.1 (from -r requirements.txt (line 5))
```
I did not face this issue on Ubuntu.. I checked the python version which is 3.6.

joamatab, 2022-05-31:
Python 3.6 has reached the end of life, why not upgrading your python using mamba or pyenv?

saicharan0112, 2022-06-01:
@joamatab it is working on Ubuntu20 distro but not in centos7. 
Anyway, I will try upgrading python too.

joamatab, 2022-06-01:
Yes, I required python 3.7 or newer

mithro, 2022-06-02:
I highly recommend using conda to install a python environment and easily get all the tools you need from https://github.com/hdl/conda-eda

saicharan0112, 2022-06-03:
@joamatab  Thank you! I found a workaround and got all of the python libraries installed.
@mithro, Thanks. I will get it a try.

closed 2022-06-03 by saicharan0112