idea-fasoc/OpenFASOC #64: This flatten way may cause the LVS check problem

Elon-J, 2022-05-30:
https://github.com/idea-fasoc/OpenFASOC/blob/b76b534c42759ce10d1a14714da1c8aca67484a1/openfasoc/common/drc-lvs-check/run_lvspex.sh#L5 @msaligane

msaligane, 2022-05-30:
Thanks @Elon-J Please let me know if it works.

saicharan0112, 2022-06-03:
Not sure if this can be helpful, but here is the PR referring to the flattening part. The CI has been failing for the flattening case - https://github.com/RTimothyEdwards/netgen/pull/45
Not clear about further details though.

msaligane, 2022-06-20:
@Elon-J Any update on this?

Elon-J, 2022-06-20:
Still need to be fixed, I think need to add a hierarchy check script

msaligane, 2022-09-05:
@Elon-J any updates on this?

Elon-J, 2022-09-06:
currently not, but we can just annotate the flatten command if we want to do the hierarchy level lvs check

saicharan0112, 2023-03-10:
@Elon-J can we close this issue? Is it relevant anymore?

saicharan0112, 2023-03-26:
I am closing this as this no longer looks relevant.

closed 2023-03-26 by saicharan0112