idea-fasoc/OpenFASOC #65: Temp-sense flow doesn't generate final GDS file with conda-eda installation

luccareinehr, 2022-06-05:
Command line log: [https://pastebin.com/gjZJYzwK](https://pastebin.com/gjZJYzwK)

I installed all dependencies (except klayout) from [https://github.com/hdl/conda-eda](https://github.com/hdl/conda-eda) repositories, then the required Python packages using `make install` from OpenFASOC's root directory. Running the temp-sense-gen with `make sky130hd_temp` fails and outputs this warning:

`FileNotFoundError: [Errno 2] No such file or directory: 'tools/../flow/results/sky130hd/tempsense/6_final.gds'`

When checking the `results/sky130hd/tempsense` folder, only these files were created:
1_1_yosys.v  2_1_floorplan.def       2_4_floorplan_macro.def    2_floorplan.def
1_synth.sdc  2_2_floorplan_io.def    2_5_floorplan_tapcell.def  2_floorplan.sdc
1_synth.v    2_3_floorplan_tdms.def  2_6_floorplan_pdn.def      2_floorplan.v

saicharan0112, 2022-06-06:
An add-on to this issue, cryo-gen is also not working

luccareinehr, 2022-08-23:
Closed by #82

closed 2022-08-23 by luccareinehr