idea-fasoc/OpenFASOC #70: Large files

aolofsson, 2022-06-24:
Perhaps not a big issue, but the repo is >800M...seems excessive. Might be worth a large file clean (saw a large mp4 in there for example): repos/OpenFASOC/tapeouts/mpw-1/testsetup/MeasResults/SanityCheckDemo.mp4

msaligane, 2022-06-25:
Thanks @aolofsson 
We will address that!

luccareinehr, 2022-06-25:
@msaligane maybe create a separate repo "openfasoc-tapeouts" for the tapeouts?

This way people who want to use openfasoc get only the tool when cloning the repository (and not other test files)

msaligane, 2022-06-26:
Good idea @luccareinehr do you want to address it?

luccareinehr, 2022-06-26:
@msaligane Yes sure, but I can't create repos in the idea-fasoc organization. 

Could you create one named openfasoc-tapeouts? Then I can submit a pull request there

msaligane, 2022-09-16:
@luccareinehr I created a new repo: https://github.com/idea-fasoc/openfasoc-tapeouts
Can you please move the files there.

luccareinehr, 2022-09-16:
@msaligane I noticed most of the remaining repository size (~400 MB) is due to the [common/platforms/](https://github.com/idea-fasoc/OpenFASOC/tree/main/openfasoc/common/platforms directory) folder. For now, I don't know what we should do about that

msaligane, 2022-09-16:
I think that could be optimized as well. Let's discuss that with @proppy

saicharan0112, 2023-03-10:
@msaligane are generators using/ planned to use any of the osu libraries listed here? [common/platforms/](https://github.com/idea-fasoc/OpenFASOC/tree/main/openfasoc/common/platforms)

msaligane, 2023-03-10:
we use them for the cryo-gen. We should find a way to have them as submodules or fetch them directly from open-pdks..

saicharan0112, 2023-03-10:
Sounds good. We can have them in a seperate repo similar to the testcases (https://github.com/idea-fasoc/openfasoc-tapeouts). Let's work on it when we get some time.

Myrausman, 2023-09-13:
I want to work on it , can you please create the repo for this [common/platforms/](https://github.com/idea-fasoc/OpenFASOC/tree/main/openfasoc/common/platforms) or share the link of it, if already made

msaligane, 2023-09-15:
@Myrausman That would be great! Here you go: https://github.com/idea-fasoc/platforms
We have a Friday weekly meeting you could join if you have cycles. Please let me know!

Myrausman, 2023-09-15:
@msaligane I'm really sorry for delayed response, thank you for creating the repo , we can do a quick meeting if you suggest.

msaligane, 2023-09-15:
Please email me at: mehdi@umich.edu