idea-fasoc/OpenFASOC #86: Can not install OpenFASOC on GCP instance-4

chenliyumich, 2022-08-24:
I tried to run (sudo) source ./Dependencies.sh on instance-4 and got this error message. It still doesn't work even with the -u option.
[image]

saicharan0112, 2022-08-24:
I tried this on ld-pll machine and it is working (except the gdsfactory installation which is to do with python version update). @msaligane @chenliyumich I am not sure which machine this is.

chenliyumich, 2022-08-24:
@saicharan0112 This is instance-4.

saicharan0112, 2022-08-24:
@chenliyumich Thanks, I still need access to this machine. I will look at it and let you know by EOD.

chenliyumich, 2022-08-24:
Thank you. Also, this is what I saw on instance-openfasoc-lcpll. Is this the machine you're working on?
[image]

saicharan0112, 2022-08-24:
Yeah. That machine is having a python version which is not supported by the gdsfactory package. It requires versions atleast 3.7. I found a workaround and it is possible to install gdsfactory on that machine. https://github.com/gdsfactory/gdsfactory/issues/627

chenliyumich, 2022-08-24:
@saicharan0112 Can you also install Magic and open_pdk globally on instance-openfasoc-lcpll? @msaligane said there's Magic installed on that machine but I could not find it.

saicharan0112, 2022-08-24:
You can install it by yourself. Remove gdsfactory in the list inside the requirements.txt file and run the Dependencies.sh script. That should do that job in this situation,

chenliyumich, 2022-08-24:
I still got this error on both instance-openfasoc-lcpll and instance-4
[image]

saicharan0112, 2022-08-24:
I think you executed this script multiple times. It already installed miniconda3 there I guess but might not be a successful installation. Can you remove the /usr/bin/miniconda3 directory and try again? `sudo ./Dependencies.sh`

chenliyumich, 2022-08-24:
I removed it and there's this message in the end of the installation. Is this neglectable? Is there a test that I can run after setup so that I know it is properly setup? Thank you
[image]

saicharan0112, 2022-08-24:
I have never seen this kind of error TBH. This error is with the centos package installer "yum". Please ping me on openfasoc channel.

saicharan0112, 2022-08-24:
@chenliyumich Also please share the entire error message. Just this screenshot doesn't help in understanding the error. This script installs magic, netgen, openroad, yosys, klayout and openroad binaries. Try searching for them using `which` command. Eg: `which magic`. The expected output is /usr/bin/miniconda3/bin/magic - if I am not wrong.

chenliyumich, 2022-08-25:
@saicharan0112 I tried "which" with the tools you mentioned. For instance-4, only Magic under /usr/local/bin/magic and Klayout under /usr/bin/klayout. For instance-openfasoc-lcpll, only klayout under /usr/bin/klayout. Also, I can't "sudo source ./Dependencies.sh" in instance-openfasoc-lcpll. It would say "sudo: source: command not found
". (But I can "sudo source ./Denpendecies.sh" in instance-4)
Is there a better way to share the entire error message? The log is very long.

chenliyumich, 2022-08-26:
@msaligane Yes. I don't know what's my next step.

msaligane, 2022-08-26:
can you try these cmds:

```
yum install m4 tcsh libX11-devel tcl-devel tk-devel ncurses-devel bzip2 mesa-libGL-devel freeglut freeglut-devel gcc-c++ python3 make git -y
git clone git://opencircuitdesign.com/magic
cd magic
./configure
make
make install
```

saicharan0112, 2022-08-26:
@msaligane can you add me to that instance 4 machine please? @chenliyumich I was able to run the script without any troubles on the pll machine.

saicharan0112, 2022-08-26:
> can you try these cmds:
> 
> ```
> yum install m4 tcsh libX11-devel tcl-devel tk-devel ncurses-devel bzip2 mesa-libGL-devel freeglut freeglut-devel gcc-c++ python3 make git -y
> git clone git://opencircuitdesign.com/magic
> cd magic
> ./configure
> make
> make install
> ```
> 
> 
> 
He already got magic installed. Weird thing is it's not from miniconda3 which is the whole purpose of the script.

saicharan0112, 2022-08-26:
@chenliyumich can you ping me on slack please?

msaligane, 2022-08-26:
@chenliyumich Please try again the lcdpll machine, I can see magic is installed there. Please confirm if you do too.

chenliyumich, 2022-08-26:
I ran the commands you provided on instance-openfasoc-lcpll, the magic is still installed at /usr/local/bin/magic. I am doing all the commands under /home/chenliyu/. For instance-4, the first line of the commands shows the error below: 
[image]
@saicharan0112 I can't find you on Slack. @msaligane Can you please add us to the same channel?

saicharan0112, 2022-08-26:
Are you on open silicon channel? The skywater ones?

chenliyumich, 2022-08-26:
@saicharan0112 No. My name on Slack is Li-Yu Chen

saicharan0112, 2022-08-26:
can you try this `sudo yum update`?

chenliyumich, 2022-08-26:
I tried it on instance-4
[image]

msaligane, 2022-08-26:
@chenliyumich You should see an invite to the openfasoc workspace.

saicharan0112, 2022-08-26:
Thanks for testing this script at your end @chenliyumich . I will try to get back to you on the issues we saw.

closed 2022-08-29 by chenliyumich