idea-fasoc/OpenFASOC #90: tempsense: manual place of header insts results in overlapping tap cells

luccareinehr, 2022-08-25:
Header cells are large enough that manual placement with the `customPlace_east` procedure makes them overlap with the tap cells in the design, failing placement checks (and thus, the flow).

(header cell highlighted)
[image]

(overlapping tap cell highlighted)
[image]

This happens when manual placement is done **before `tapcell` is called and also after**.

msaligane, 2022-08-26:
We should have multiple versions of these header cells (different sizing, and different layouts). That would increase DSE options and probably improve results. 

Another point is that we could use differential inverters in the subthreshold ROs. Or other topologies.

luccareinehr, 2022-08-27:
Related: https://github.com/The-OpenROAD-Project/OpenROAD/issues/2210

luccareinehr, 2022-10-18:
Closed by #111

closed 2022-10-18 by luccareinehr