idea-fasoc/OpenFASOC #97: git ignore generated verilog files

luccareinehr, 2022-09-12:
Generated verilog files from the generators shouldn't be uploaded to git, only the templates should.

In the temp-sense-gen, for example, this is currently happening: in [temp-sense-gen/src/](https://github.com/idea-fasoc/OpenFASOC/tree/main/openfasoc/generators/temp-sense-gen/src), `TEMP_ANALOG_hv.nl.v` and `TEMP_ANALOG_lv.nl.v` are uploaded to GitHub, but they are generated files. Take a look:

In `TEMP_ANALOG_hv.nl.v`:
[image]
These lines depend on the number of HEADER cells chosen, which is an input of the generator.

For the temp-sense-gen, it's an easy fix I can make. But does this also happen in other generators?

luccareinehr, 2022-09-12:
In temp-sense-gen, there are also generated files in https://github.com/idea-fasoc/OpenFASOC/tree/main/openfasoc/generators/temp-sense-gen/flow/design/src/tempsense

saicharan0112, 2022-10-31:
@luccareinehr This can be closed right? You mentioned it in the above 2 PRs

luccareinehr, 2022-10-31:
For the tempsense yes, but I don't know what's the situation on other
generators... afaik the ldo-gen still uploads some files it generates for
instance

saicharan0112, 2022-10-31:
let's close this and ask pranav to consider this when he creates his PR. 

@lulu9312 Do you understand the problem Lucca is addressing here?

closed 2022-10-31 by saicharan0112