idea-fasoc/OpenFASOC #109: tempsense: lvs stopped working after openroad pull #2289

luccareinehr, 2022-10-07:
https://github.com/The-OpenROAD-Project/OpenROAD/pull/2289/files

To do:
* Modify pdngen to pdn.tcl
* Fix route not getting created from header cells to VIN
* Check if LVS is now not giving mismatches

saicharan0112, 2022-10-07:
https://github.com/The-OpenROAD-Project/OpenROAD/pull/2289 - To track from the openroad repo side.

luccareinehr, 2022-10-10:
A note to myself: LVS won't work if make is run from ORFS flow/Makefile. It needs to be run from OpenFASOC's tempsense Makefile

luccareinehr, 2022-10-10:
To [document the flow](https://openfasoc.readthedocs.io/en/latest/flow-tempsense.html#rtl-to-gds-flow) after #87, it was basically needed to (re)create a second net (r_VIN) that makes the route from the ring, the net's block terminal (BTerm), to the HEADER cells.

Otherwise, if we just tweaked the original VIN net, all instances in the stdcell grid would get routed to it...

PR #111 (2022-10-12): tempsense: fix LVS, pdn.cfg -> pdn.tcl, move custom header placement to its own file
- Closes #109

- Closes #81

- Moves the custom placement procedure from #77 to a separate script, and automatically rotates the placed cells according to their row direction (to align correctly with VDD / GND)

- Adds an option in `customPlace_east` to allow the custom placed instances to move slightly during detailed placement, to avoid overlaps with other fixed cells (tap cells for example)

- Also addresses #97 for the temp-sense-gen (for other generators, the issue might still be valid)
closed 2022-10-12 by msaligane