idea-fasoc/OpenFASOC #114: tempsense: fail run when it encounters LVS errors

luccareinehr, 2022-10-18:
For some types of LVS errors, `netgen` does not exit with `return code 1` and the temp-sense-gen still says that "LVS is clean!" when it is not.

For example:
[image]

Here I highlight the line printed by netgen that's attesting the LVS error. Notice how at the end of the generator run it still says that there was no error.

That is also the reason why sometimes the temp-sense-gen CI doesn't fail even with some LVS mismatches.

saicharan0112, 2022-10-18:
@luccareinehr good one! So we need an effective lvs report parser?

saicharan0112, 2022-10-18:
Or should this be diverted to Tim Edwards to take care of this inside Netgen itself?

luccareinehr, 2022-10-18:
@saicharan0112 Good idea, ideally it should be changed in netgen. If they disagree then we can make a parser

saicharan0112, 2022-11-01:
@luccareinehr is this still seen in the lvs report? Can we provide the testcase by packing the latest netlists to Tim Edwards to ask if he can take care of this inside Netgen?

msaligane, 2022-12-02:
@luccareinehr is this ready to close?

saicharan0112, 2023-01-01:
@luccareinehr Its just that the report parser python script present inside the tools directory has hardcoded lvs error message. I just modified it to catch any type of "fail" messages. Not sure if you had any conversation with Tim Edwards on this.

Happy New year BTW ;)

commit 259daeb9 (2023-01-06): modify lvs parser to catch failed error msgs - fixes #114
closed 2023-01-01 by saicharan0112