idea-fasoc/OpenFASOC #132: tempsense: avoid metal 1 in the VIN route

luccareinehr, 2022-11-12:
Lower metal layers have higher resistivity, so we should ignore them for power routing

msaligane, 2022-11-12:
Lower metal layer should not be used for routing these special nets. We also have to find a way to avoid single via connection.
We really need a Special router or point-to-point router.

luccareinehr, 2022-11-12:
We can set the top and bottom routing layers with the detailed_place command

So we can first route all nets except r_VIN, then just route r_VIN limiting the layers

msaligane, 2022-11-12:
Yes, that should work. But at this point, we have too many hacks to do something so simple.
We need to create a specification so that OpenROAD developers add it.

saicharan0112, 2023-03-22:
@luccareinehr any plans to work on this?

luccareinehr, 2023-03-23:
@saicharan0112 not really, sorry... I remember all I could try were some "hacks" with openroad commands, which eventually became too complicated. Hence the need for the special router Mehdi mentions

saicharan0112, 2023-03-24:
No problem! Let's see if anyone could pick this up.