idea-fasoc/OpenFASOC #135: Some problems about  LDO generation

CaoChengwei9, 2022-11-13:
Hi, I have installed this repository following `getting-started.rst`.Then I came to the directory of `OpenFASOC/openfasoc/generators/ldo-gen`, and I entered the  command `make` but I got an error as followed:

  `File "./tools/ldo_gen.py", line 292, in <module>`
 `   coefLength = len(jsonModel["Iload,max"][str(vin)])`
`KeyError: '1.8'`
`make: *** [Makefile:2: sky130hvl_ldo_verilog] Error 1`

How can I solve this problem?
Also, I want to know how to generate DLDO from RTL to GDSII, just like your demo shows on the [website](https://fasoc.engin.umich.edu/digital-ldo/). However, I have no idea about the usage of ldo-gen.

Hope for answers.Thanks

msaligane, 2022-11-13:
@lulu9312
can you please take a look?

lulu9312, 2022-11-13:
Hello @CaoChengwei9 , the make command will throw this error because the model file is not updated yet. It is work in progress. For generation of RTL to GDSII , you can run make finish by going into  OpenFASOC/openfasoc/generators/ldo-gen/flow. To view final result you need run make gui_final once the flow is complete.

Hope this helps. Thanks

lulu9312, 2022-11-13:
> @lulu9312 can you please take a look?

@msaligane  Yes. Can you assign me to this issue? I am not able to assign it.

CaoChengwei9, 2022-11-13:
It does work!
But after entering `make gui_final`,I meet this error:
`[ERROR] This code was compiled with the GUI disabled.  Please recompile with Qt5 if you want the GUI.`
Is this my question or something else?

Sincerely thanks!

saicharan0112, 2022-11-13:
Is that openroad error?

lulu9312, 2022-11-13:
@CaoChengwei9 You need to install qt5. You can install it by running sudo apt-get install qt-5 default And then rebuild Openroad again. This should resolve this error. 
@saicharan0112 Yes this is OpenROAD error.

CaoChengwei9, 2022-11-13:
@lulu9312 Thank you.I have checked that qt5 has already been installed:
`qt5-default is already the newest version (5.12.8+dfsg-0ubuntu2.1).`
`0 upgraded, 0 newly installed, 0 to remove and 20 not upgraded.`
But the error is not solved.Why would this error happen?

saicharan0112, 2022-11-13:
@CaoChengwei9 Did you install using the ./dependencies.sh script that is shown in the `getting-started.rst` page?

CaoChengwei9, 2022-11-13:
@saicharan0112 Yes. I follow that page and use `./dependencies.sh` command.
[image]
I enter the command `openroad` and no gui appears. Is that problem with openroad?

saicharan0112, 2022-11-13:
I thought so. The way that script installs the openroad is using its conda
package distribution from conda-eda (Litex-hub anaconda repo). And that
installation is without a ***@***.*** <https://github.com/lulu9312> This
flow doesn't work without a GUI?

CaoChengwei9, 2022-11-13:
@saicharan0112 I think it may be so. If I couldn't view final results via GUI, how to custom other parameters like Vin, Iload,max ?
So I would try the manual installation of OpenROAD for GUI.
Thanks a lot.

saicharan0112, 2022-11-13:
If you just want to observe the GDS, you can do so by opening the final
6_final.gds using Klayout. That script actually downloads the Klayout using
the distributed debian package. Also if you are manually installing
openroad, make sure you remove the miniconda3 path from the $PATH variable.
Else you might end up in accessing the conda version of openroad since that
has precedence over your /usr/bin or /usr/local/bin or /usr/share/bin paths
of accessing binary files.

saicharan0112, 2022-11-13:
@lulu9312 <https://github.com/lulu9312> can't we have a batch version of
the flow, something like a temp-sense generator?

msaligane, 2022-11-14:
@CaoChengwei9 The issue you are seeing is related to OpenROAD's build. You need to rebuild with QT5, as the error messages show.

What kind of application or spec. you are looking for? I am to provide feedback on your system.

CaoChengwei9, 2022-11-14:
@msaligane Thanks for your answer.
I'm looking for large load current and generation of customed auxiliary cells. 
For example, load current of single-DLDO on my system would be almost 70mA or higher.
Besides, I want to replace the comparator and controller in the prototype with my own design.

msaligane, 2022-11-16:
@CaoChengwei9 Sounds interesting! Which technology are you using? I am assuming SKY130?
If so, would you be willing to push your variants to the repo?
Please email me at: mehdi@umich.edu

lulu9312, 2023-02-17:
@CaoChengwei9 I think this issue has been resolved. Closing this now

closed 2023-02-17 by lulu9312