idea-fasoc/OpenFASOC #187: Generated Files Are Not Gitignored In Some Generators

harshkhandeparkar, 2023-03-06:
Generated files such as `.log`, `.rpt`, `.odb` in `flow/`, `work/`, and `simulations/run/` are not gitignored for the following generators:
1. LDO: No files are gitignored
2. Temperature  Sensor: `simulations/run/` is not gitignored.

harshkhandeparkar, 2023-03-06:
I was going through the documentation and came across this while running the LDO generator.
I am currently learning about the tool and hoping to contribute in the future so may I work on a fix for this?

harshkhandeparkar, 2023-03-09:
I also found that some generated files which were not gitignored are tracked. Should I also untrack these files?

Example: Generated verilog files in ldo-gen and dcdc-gen
- [LDO_CONTROLLER.v](https://github.com/idea-fasoc/OpenFASOC/blob/b9784507aac426970b56492f8327033e1a4feb15/openfasoc/generators/ldo-gen/flow/design/src/ldo/LDO_CONTROLLER.v)
- [DCDC_BUFFER.sv](https://github.com/idea-fasoc/OpenFASOC/blob/b9784507aac426970b56492f8327033e1a4feb15/openfasoc/generators/dcdc-gen/src/DCDC_BUFFER.sv)

saicharan0112, 2023-03-10:
@harshkhandeparkar I don't think excluding the folders and files you mentioned is a good idea. Intended users are designers and usually these files should be under version control

lulu9312, 2023-03-10:
@saicharan0112 The folders that are generated such as logs/ results/ will change everytime the generator is run. So I think those files should not be pushed to github because anyway when running the flow again the files will be created again.

harshkhandeparkar, 2023-03-10:
> @harshkhandeparkar I don't think excluding the folders and files you mentioned is a good idea. Intended users are designers and as far as I know, usually they want these files under version control

There are templates (which are pushed) for generating these files and they are generated everytime the generator is run. Also these files change slightly depending on the configuration. Also there are nearly a hundred files that are generated so I don't think pushing these is a good idea.

saicharan0112, 2023-03-10:
> I was going through the documentation and came across this while running the LDO generator. I am currently learning about the tool and hoping to contribute in the future so may I work on a fix for this?

I really appreciate your interest. Can you look at this issue? It would be really wonderful if you can help with this? https://github.com/idea-fasoc/OpenFASOC/issues/37

closed 2023-03-22 by saicharan0112