idea-fasoc/OpenFASOC #196: Klayout LVS doesn’t extract floating N-FET body tap

StanleyAlwaysLoveYou, 2023-03-27:
# Expected Behavior

The Klayout should extract all four terminals of the NFET layout when running LVS

# Actual Behavior 

No body terminal shows in the extracted netlist from klayout

# Reproducing process

- The layout of 1 PFET and 1NFET

[image]

- Extraction result from Klayout, note the lack of “NBODY”.  Instead, Klayout assumes that all NFET bodies are tied to the global ground node.

[image]

# Affect

This can lead to potential latch-up. Cadence flagged this as a soft check error.

StanleyAlwaysLoveYou, 2023-03-27:
@andylithia @msaligane FYI

msaligane, 2023-03-29:
Have you opened a Github issue for this?
@proppy @mithro