idea-fasoc/OpenFASOC #204: Make ldo generator simulation processing part similar to temp-sense generator

saicharan0112, 2023-04-09:
@lulu9312 Currently the processing part after simulations are not modular. I can't reuse it for regression without depending on different variables and functions inside the processing.py script. Can you please keep everything inside one script so that the entire processing part is handled by just running that script?

Please refer to temp-sense generator for clear understanding.

saicharan0112, 2023-04-09:
I don't think this will take much time. Do you think you can do this by Monday?

lulu9312, 2023-04-09:
> I don't think this will take much time. Do you think you can do this by today?

@saicharan0112 I can get this done by tomorrow.

saicharan0112, 2023-05-01:
https://github.com/idea-fasoc/OpenFASOC/pull/202 solves this.

closed 2023-05-01 by saicharan0112