idea-fasoc/OpenFASOC #207: Colab fails to install gdstk

seijirom, 2023-04-21:
During the first step (install all tools and dependencies), colab fails to pip install gdstk as shown below:

Collecting gdstk
  Downloading gdstk-0.9.37.tar.gz (509 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 509.2/509.2 kB 13.1 MB/s eta 0:00:00
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Requirement already satisfied: numpy in /usr/local/lib/python3.9/dist-packages (from gdstk) (1.22.4)
Building wheels for collected packages: gdstk
  error: subprocess-exited-with-error
  
  × Building wheel for gdstk (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> See above for output.
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  Building wheel for gdstk (pyproject.toml) ... error
  ERROR: Failed building wheel for gdstk
Failed to build gdstk
ERROR: Could not build wheels for gdstk, which is required to install pyproject.toml-based projects`

seijirom, 2023-04-21:
Thank you! It worked! Completed the whole Jupyter Notebook.

saicharan0112, 2023-04-21:
@seijirom can you send a PR updating this inside the notebook

saicharan0112, 2023-05-01:
@seijirom wanna send a PR?

seijirom, 2023-05-02:
@saicharan0112 I've tried but gave up because updating code in ipynb file in my environment might cause a trouble.
I  hope someone in the team will fix the ipynb file.
Thank you.

closed 2023-09-11 by saicharan0112