idea-fasoc/OpenFASOC #208: PnR failed for mixed signal design, signal 11 received during routing

rahulearn2019, 2023-04-22:
While trying to generate the gds, i am getting following error during global route stage
It says route not created for one of the inputs to design and then says - "signal 11 received" PnR failed

[5_1_fastroute.log](https://github.com/idea-fasoc/OpenFASOC/files/11301113/5_1_fastroute.log)

rahulearn2019, 2023-04-22:
[image]

saicharan0112, 2023-04-22:
Can you give more info on how this is relevant to OpenFASOC, please? I see
that you are using openroad flow scripts github repository.

rahulearn2019, 2023-04-22:
The design has a analog block whose GDS is generated using ALIGN and provided as dummy verilog to the flow. It has a mux as the digital block. OpenFASOC uses OpenROAD for RTL2GDS right? OpenROAD Flow takes a design from the IMC-gen/flow/design/ directory and runs it through its flow, generating a DEF and a GDS at the end. The design is specified by using the generated Verilog files and a config.mk file that configures OpenROAD Flow to the design

saicharan0112, 2023-04-22:
How are you using OpenFASOC for your analog block here? It would be helpful if you are more specific with OpenFASOC.

saicharan0112, 2023-09-11:
Not sure if this is relevant anymore so closing it.

closed 2023-09-11 by saicharan0112