idea-fasoc/OpenFASOC #209: temp-sense generator fails with the latest tools

saicharan0112, 2023-05-08:
The temperature sensor generator is having drc and lvs errors with the latest dependencies. Attached the `work` folder containing the reports and the final gds file.
[work.tar.gz](https://github.com/idea-fasoc/OpenFASOC/files/11422242/work.tar.gz)

saicharan0112, 2023-05-08:
https://github.com/idea-fasoc/OpenFASOC/blob/main/versions.txt
https://github.com/idea-fasoc/OpenFASOC/blob/main/conda_versions.txt

michahn01, 2023-05-09:
The current official version of OpenFASOC has a syntax error in dependencies.sh. The `if` block that is opened in Line 26 is not closed. This causes an "unexpected end of file" error after Line 26. When i ran this on docker, the script stopped at Line 26 and didn't download anything that comes after (i.e., ngspice, klayout, xyce).

michahn01, 2023-05-10:
> @michahn01 I don't think these both are related ? doesn't your PR majorily overwrites this file?

From what I've tested on Docker, installing OpenFASOC and running tempsensegen with my PR produces no error (clean DRC and LVS). For reference, here is the Docker file test case I used:

`FROM ubuntu:20.04

RUN apt-get -y update
RUN apt-get -y upgrade
ENV DEBIAN_FRONTEND=noninteractive
RUN apt-get install -y wget git python3 vim
RUN git clone https://github.com/michahn01/OpenFASOC.git
RUN cd OpenFASOC
RUN ./dependencies.sh
RUN export PATH=/usr/bin/miniconda3/bin:$PATH
RUN export PDK_ROOT=/usr/bin/miniconda3/share/pdk/`

michahn01, 2023-05-11:
@saicharan0112 

I think the issue is with open_pdks.sky130a specifically. Using the latest version of any other file does not result in the error if open_pdks.sky130a is set to the safe version. The error happens only when using the latest version of open_pdks.sky130a.

saicharan0112, 2023-05-11:
Yes I am already aware of that. Do you anything about those drc and lvs
errors?

michahn01, 2023-05-15:
Actual latest versions of dependencies:
Netgen: v1.5.253
Ngspice: v36
Klayout: v0.28.7
Yosys: v0.29+11
OpenROAD: v2.0-7559-gdf3bfc5a5
**Open_pdks: v1.0.414**
Xyce: v7.6
MAGIC: v8.3.398

Latest versions of packages available on LiteX-Hub:
Netgen: v1.5.251 (on Linux)
Ngspice: v39.3
Klayout: v0.28.6 (on Linux)
Yosys:  v0.27 (on Linux)
OpenROAD: v2.0_7681_g850238c2f (on Linux)
**Open_pdks:  v1.0.406**
MAGIC:  v8.3.389

closed 2023-09-11 by saicharan0112