idea-fasoc/OpenFASOC #210: Plan enhancements of gdsfactory based generators

alibillalhammoud, 2023-05-12:
The following are the big action items @vijayshankarr and I plan to work on to enhance the gdsfactory based generators. This issue will be updated with PR addressing each point and with relevant discussion. 

- [ ]  Organize file structure of the gdsfactory generators:   
Currently there are several directories with generator codes; we will consolidate these into the gdsfactory-gen folder under "generators". The other folders will be deleted once this has been accomplished.

- [ ]  Introduce additional pcells as highly versatile python modules:   
We will develop the additional pcells as individual modules. Modules can be easily imported and will be as generic as possible so that cells can be instantiated with custom parameters. We plan to start with sky130 pdk, but will add support for additional technologies while reusing as much code as possible.

- [ ]  Automatic Scripts:  
Create a DRC script that can be used in regression testing.  
Create an extraction script to dump netlist from GDS (may be required to use pcells in other designs and run simulations).

The items presented are long term goals and we will include incremental updates as we decide how to best achieve these goals.

alibillalhammoud, 2023-06-14:
Additional pcells as highly versatile python modules:
We can validate layers with GDSFactory when passing PDK to ComponentFactory. See GDSFactory [issue1731](https://github.com/gdsfactory/gdsfactory/issues/1731) and merged [PR1761](https://github.com/gdsfactory/gdsfactory/pull/1761)

vijayshankarr, 2023-06-15:
@alibillalhammoud did you have any update on the Pdk.activate() bug from GDSFactory?

alibillalhammoud, 2023-06-22:
@vijayshankarr We can add the rectangular ring we use in the fet generators to GDSFactory. I will create an issue in GDSFactory and get feedback on it.

alibillalhammoud, 2023-07-10:
Merged from [Vijay](https://github.com/idea-fasoc/OpenFASOC/pull/189) fixes folder organization

alibillalhammoud, 2023-07-17:
[218](https://github.com/idea-fasoc/OpenFASOC/pull/218) from myself and @ryanrocket will close this issue

closed 2023-09-26 by alibillalhammoud