idea-fasoc/OpenFASOC #228: installation procedure

ripudamank2, 2023-08-22:
Hi,
"I already have Magic, Klayout, Yosys, Openroad, and open_pdk installed on my Linux machine. How can I install this new tool without affecting my existing tools?"

saicharan0112, 2023-08-23:
@ripudamank2 I would suggest you to start with the documentation. https://openfasoc.readthedocs.io/ Its merely cloning this repo and running the generates using the provided makefile provided under the generator

ripudamank2, 2023-08-25:
i think your not getting my point . I already has those tool which you mentioned with OpenFASOC and i dont want to install again . 
What the procedure to just setup FASOC without installing those tool .

msaligane, 2023-08-25:
Then you just have to run the generators and pray you have the right versions. 

The doc shared by Sai describes how to run it. Cheers.

closed 2023-08-25 by msaligane

saicharan0112, 2023-08-25:
> i think your not getting my point . I already has those tool which you mentioned with OpenFASOC and i dont want to install again . What the procedure to just setup FASOC without installing those tool .

I still stand by my answer. Let me break it down to you. Follow the docs, check the versions supported, install conda and those respective conda versions, clone the OpenFASoC git repo, go back to the doc and read how to run the generator you are interested in and follow those instructions.