idea-fasoc/OpenFASOC #231: Segregate Test Workflows by Paths

harshkhandeparkar, 2023-08-23:
Currently, all testing workflows are run on creating a pull request.
https://github.com/idea-fasoc/OpenFASOC/blob/cdcd24ff1e8d20e2994cfe88b7b71ba4f7d20e27/.github/workflows/tempSense_sky130hd.yml#L3-L8

This is both inefficient and unnecessary. It would be better if paths are [specified](https://docs.github.com/en/actions/using-workflows/workflow-syntax-for-github-actions#onpushpull_requestpull_request_targetpathspaths-ignore) in the workflow events so that only the affected generators are tested in a pull request.

harshkhandeparkar, 2023-08-23:
> @harshkhandeparkar how do you know which generators are affected because of changes?

Some paths would affect all generators - `common`, for example. But generators should not interfere with each other. So we could run the tests when any common path is changed or any file under the generator's directory is changed.

saicharan0112, 2023-08-23:
sounds good. Can you modify the yml files accordingly?

harshkhandeparkar, 2023-08-23:
> sounds good. Can you modify the yml files accordingly?

Sure. I'll open a PR soon.