idea-fasoc/OpenFASOC #235: Gdsfactory-gen Improvments

alibillalhammoud, 2023-09-05:
Todos for improving gdsfactory-gen code. Feel free to add more items here: 

- ReadME enhancements for code in the pygen.common directory
- rename pygen
- organize gds files in tapeout_and_RL directory
- add discussed pcells under pygen.common directory
- import klayout drc code from pdk (.drc script)
- add documentation content to the readthedocs website
- publish package to pypi

alibillalhammoud, 2023-09-05:
@saicharan0112 @msaligane @alibillalhammoud

alibillalhammoud, 2023-09-11:
Added comments from @harshkhandeparkar

harshkhandeparkar, 2023-09-12:
> follow up on requirements.txt: GDSFactory, sky130, and GF180 pypi packages cannot be installed at the same time without broken deps

Elaborating on this: `gf180` and `sky130` have different version requirements of gdsfactory. sky130 requires `v7.3.0` and gf180 requires `v7.1.4`. When either package is installed, it replaces the previously installed version of gdsfactory, leading to broken dependencies. The following error is displayed on installing sky130 after gf180.

```
Successfully uninstalled gdsfactory-7.1.4
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
gf180 0.0.3 requires gdsfactory[cad]<7.2.0,>=7.1.1, but you have gdsfactory 7.3.0 which is incompatible.
Successfully installed gdsfactory-7.3.0
```

Even though this error is displayed, the generators still seem to work.

harshkhandeparkar, 2023-09-12:
Opened an issue in the relevant gdsfactory repo: https://github.com/gdsfactory/skywater130/issues/73.

alibillalhammoud, 2024-06-05:
I am closing this issue and creating a new one as these tasks have been completed or deprecated

closed 2024-06-05 by alibillalhammoud