idea-fasoc/OpenFASOC #241: LDO-gen make fails after LVS and DRC

chetanyagoyal, 2023-09-25:
The make command, executed as follows 
`make sky130hvl_ldo`
fails with the following error message 
```#----------------------------------------------------------------------
# LVS and DRC finished successfully
#----------------------------------------------------------------------
Traceback (most recent call last):
  File "./tools/ldo-gen.py", line 188, in <module>
    rawPEXPath, jsonConfig["simTool"],user_specs["designName"]
  File "/home/chetanya/OpenFASOC/openfasoc/generators/ldo-gen/tools/simulations.py", line 118, in process_PEX_netlist
    vref_node_to = re.findall(r"\bcapacitor_test_nf_\S*", netlist)[0]
IndexError: list index out of range
make: *** [Makefile:57: sky130hvl_ldo] Error 1
```

For reference, temp-sense-gen and cryo-gen execute to completion
This error has been replicated on another system, requires fixing

saicharan0112, 2023-10-01:
@harshkhandeparkar isn't this something you are handling already?

harshkhandeparkar, 2023-10-01:
> @harshkhandeparkar isn't this something you are handling already?

Yes. Chetanya contacted me and we are trying to figure it out.

chetanyagoyal, 2023-10-07:
> > @harshkhandeparkar isn't this something you are handling already?
> 
> Yes. Chetanya contacted me and we are trying to figure it out.

I will try and integrate @harshkhandeparkar and @saicharan0112's forks to solve this problem for the latest edition of tools

saicharan0112, 2023-10-10:
@harshkhandeparkar Is this issue fixed in one of the PRs that were merged recently? Also can you create a new issue on PnR step failing for LDO please?

harshkhandeparkar, 2023-10-10:
> @harshkhandeparkar Is this issue fixed in one of the PRs that were merged recently? Also can you create a new issue on PnR step failing for LDO please?

Yes this was fixed. I haven't replicated the issue on my system yet. @chetanyagoyal could you open the issue?

closed 2023-10-10 by saicharan0112