idea-fasoc/OpenFASOC #248: Install Pypi Modules in the Docker Image

harshkhandeparkar, 2023-10-09:
The docker image `msaligane/openfasoc` includes all the conda dependencies but does not install the Python/Pypi dependencies listed in the `requirements.txt`, and they have to be installed separately inside the container. It would be useful to add them to the built image.

saicharan0112, 2023-10-09:
That image is used to test generators and the `requirements.txt` file is part of the generators development (people add libraries that are used in these generators) and it is important to test them regularly. you can ofcourse create your own image using this base image.

harshkhandeparkar, 2023-10-09:
> That image is used to test generators and the `requirements.txt` file is part of the generators development (people add libraries that are used in these generators) and it is important to test them regularly. you can ofcourse create your own image using this base image.

Aren't the libraries required for testing as well?

saicharan0112, 2023-10-09:
> Aren't the libraries required for testing as well?

Of course. That's why we first test them by installing them before we test generators

harshkhandeparkar, 2023-10-09:
> Of course. That's why we first test them by installing them before we test generators

Why do they have to be installed separately? Can't they be installed in the build step itself since the image is also updated regularly?

saicharan0112, 2023-10-09:
We build the image once in a day and the generator can be updated multiple times. Requirements file is part of the generators bundle. Checks are done on the generators hence we load both the python libraries list and generators code. 

What exactly is your usecase @harshkhandeparkar ?

harshkhandeparkar, 2023-10-10:
> We build the image once in a day and the generator can be updated multiple times. Requirements file is part of the generators bundle. Checks are done on the generators hence we load both the python libraries list and generators code.
> 
> What exactly is your usecase @harshkhandeparkar ?

I use it for development as well. Since many of the libraries will not change so we can pip install in the Dockerfile as well as in the CI workflow and it will save some time after pulling the container since some deps will already be installed.

Doing this will be better for development since pip install won't have to be run in most cases for that.

saicharan0112, 2023-10-10:
we can have one more version of this for testing. That should be no prob.

saicharan0112, 2023-10-10:
but if you want to open the openroad gui or the klayout while debugging, it is not possible. From my past experiences, no one was using the docker images. everyone is working from the GCP VM and all tools are installed there already. I think you are the only person (and probably Chetanya) want this one.

harshkhandeparkar, 2023-10-10:
> but if you want to open the openroad gui or the klayout while debugging, it is not possible. From my past experiences, no one was using the docker images. everyone is working from the GCP VM and all [...]

Which VM instance are you talking about? The docker image is quite useful for development since I can run it locally. It would be cool if we could publish a development docker image that would help new contributors get started without having to install or compile the tools themselves.

closed 2023-10-11 by saicharan0112