idea-fasoc/OpenFASOC #249: Cryo-gen fails at placement stage with the new openroad version

saicharan0112, 2023-10-10:
Cryo-gen fails at placement stage with the latest openroad version - https://github.com/idea-fasoc/OpenFASOC/actions/runs/6464762054/job/17549890129#step:3:3064

saicharan0112, 2023-11-11:
```
[ERROR DPL-0039] "sky130_fd_sc_hs__decap_4" did not match any masters.
This could be due to a change from using regex to glob to search for cell masters. https://github.com/The-OpenROAD-Project/OpenROAD/pull/3210
Error: detail_place.tcl, 20 DPL-0039
```

chetanyagoyal, 2023-12-22:
[https://github.com/The-OpenROAD-Project/OpenROAD/pull/3210](url) This change causes the script to error out if the platform doesn't match, where it would have previously just thrown a soft warning

chetanyagoyal, 2023-12-22:
Once this is fixed, the flow further errors out in `final_report.tcl` in
```tcl
if {[expr [llength [info procs save_image]] > 0]} {
    gui::show "source $::env(SCRIPTS_DIR)/save_images.tcl" false
}
```
^ this statement. The `scripts` folder does not seem to have a `save_images.tcl`, which seems like enough to conclude that we don't want the functionality from this last statement, so I think it can be commented out.

closed 2024-02-14 by chetanyagoyal