idea-fasoc/OpenFASOC #253: enable simulations in the CI

saicharan0112, 2023-10-18:
- [ ] Update docker image `msaligane/openfasoc` to include ngspice
- [ ] Pick config info for each generator and store the required output files for comparison
- [ ] Build scripts to run the comparison after the run inside the workflow

@chetanyagoyal

saicharan0112, 2023-11-08:
@chetanyagoyal please add the PR links beside the tasks - feel free to elaborate the tasks with sub-tasks if needed. Refer to PRs from Harsh and Ali - https://github.com/idea-fasoc/OpenFASOC/issues/211 etc. These are good practices.

chetanyagoyal, 2023-11-08:
* [x]  Update docker image `msaligane/openfasoc` to include ngspice
 * [ ] Pick config info for each generator and store the required output files for comparison
* * [ ] Run simulations for all operating temperatures
* * [ ] Store output files in subdirectory in  .github/scripts
* * [x] Propogate out the simulation state from python simulation scripts
 * [x]  Build scripts to run the comparison after the run inside the workflow
* [x] Implement a new yaml file to run the temp-sense simulation workflow
* [ ] Clean up

saicharan0112, 2023-11-09:
@chetanyagoyal please reuse the same workflow file for temp-sense to run the simulations. please avoid creating multiple workflows for a single generator