idea-fasoc/OpenFASOC #254: problem while running make sky130hd_temp

ripudamank2, 2023-10-22:
finish report_design_area
--------------------------------------------------------------------------
Design area 2551 u^2 19% utilization.

[ERROR GUI-0013] Unable to find visible display control at Pin Markers.
[ERROR GUI-0070] Error: save_images.tcl, 23 GUI-0013
Error: final_report.tcl, 63 GUI-0070
Command exited with non-zero status 1
Elapsed time: 0:00.67[h:]min:sec. CPU time: user 0.54 sys 0.11 (97%). Peak memory: 226936KB.
make[1]: *** [Makefile:568: /home/embedded/Documents/OpenFASOC/openfasoc/generators/temp-sense-gen/flow/logs/sky130hd/tempsense//6_report.log] Error 1
make[1]: Leaving directory '/home/embedded/Documents/OpenFASOC/openfasoc/generators/temp-sense-gen/flow'
[Error] Place and Route failed. Refer to the log file
make: *** [Makefile:48: sky130hd_temp] Error 1

question : 
1. Whats the use of Xyce and trilinos ?
2. How I can solve above problem ?

chetanyagoyal, 2023-10-26:
@ripudamank2 to see how to fix the problem, you can go through the changes made in the PR that has been linked

chetanyagoyal, 2023-10-30:
@ripudamank2 To fix this error, in the file save_images.tcl (in flow/scripts in the temp-sense-gen folder).  change "Pin Markers" in line 23 to "Pins"

ripudamank2, 2023-10-31:
I did that and now getting new error . Please check below open issue : 
 **error LVS finished #257**

saicharan0112, 2023-11-03:
@ripudamank2 I am closing this issue since you have already opened a new issue for the lvs, Thanks for reporting!

PR #256 (2023-11-03): Update save_images.tcl
Fixes #254.
Updates the save_images.tcl file to accommodate for the changes made in Openroad-flow-scripts
closed 2023-11-03 by saicharan0112