idea-fasoc/OpenFASOC #257: error LVS finished

ripudamank2, 2023-10-26:
xtracting sky130_fd_sc_hd__or3b_2 into sky130_fd_sc_hd__or3b_2.ext:
Extracting tempsenseInst_error into tempsenseInst_error.ext:
tempsenseInst_error: 2 warnings
Total of 10 warnings.
exttospice finished.
optfile ./ng.opt does not exist - using default values
togl-version : 2
OCC module loaded
Thank you for using NETGEN
make[1]: Leaving directory '/home/embedded/Documents/OpenFASOC/openfasoc/generators/temp-sense-gen/flow'
#----------------------------------------------------------------------
# LVS finished
#----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/embedded/Documents/OpenFASOC/openfasoc/generators/temp-sense-gen/tools/temp-sense-gen.py", line 223, in <module>
    shutil.copyfile(
  File "/usr/lib/python3.10/shutil.py", line 254, in copyfile
    with open(src, 'rb') as fsrc:
FileNotFoundError: [Errno 2] No such file or directory: 'tools/../flow/reports/sky130hd/tempsense/6_final_lvs.rpt'
[image]

make: *** [Makefile:48: sky130hd_temp] Error 1

what should i do ? I am attaching screenshot also .

saicharan0112, 2023-11-07:
@ripudamank2 can you please share the entire log file for complete understanding on what's happening at your side? Usually (probably not always) an error at this is because of something failing at PnR stage.

hareyakana, 2023-11-08:
i have similar error. i think the issue is from line 185-226 in temp-sense-gen.py. Python's shutil is not handling the long args well and it is ignored entirely by python and it is not excuted.

ripudamank2, 2023-11-09:
Updated 0 paths from the index
sed: couldn't open temporary file /usr/local/share/pdk//sky130A/libs.tech/magic/sed5azpYH: Permission denied
CIF file read warning: CIF style sky130(): units rescaled by factor of 5 / 1
Error while reading cell "tempsenseInst_error" (byte position 524262): Unknown layer/datatype in boundary, layer=14 type=0
CIF file read warning: CIF style sky130(): units rescaled by factor of 5 / 1
Error while reading cell "tempsenseInst_error" (byte position 524262): Unknown layer/datatype in boundary, layer=14 type=0
Total of 10 warnings.
Total of 10 warnings.
Total of 10 warnings.
Traceback (most recent call last):
  File "/OpenFASOC/openfasoc/generators/temp-sense-gen/tools/temp-sense-gen.py", line 223, in <module>
    shutil.copyfile(
  File "/usr/lib/python3.10/shutil.py", line 254, in copyfile
    with open(src, 'rb') as fsrc:
FileNotFoundError: [Errno 2] No such file or directory: 'tools/../flow/reports/sky130hd/tempsense/6_final_lvs.rpt'
make: *** [Makefile:48: sky130hd_temp] Error 1
[test_loop.txt](https://github.com/idea-fasoc/OpenFASOC/files/13312681/test_loop.txt)

Please check the log and terminal output

saicharan0112, 2023-11-11:
> **_Updated 0 paths from the index sed: couldn't open temporary file_** **_/usr/local/share/pdk//sky130A/libs.tech/magic/sed5azpYH: Permission denied_** CIF file read warning: CIF style sky130(): units rescaled by factor of 5 / 1 Error while reading cell "tempsenseInst_error" (byte position 524262): Unknown layer/datatype in boundary, layer=14 type=0 CIF file read warning: CIF style sky130(): units rescaled by factor of 5 / 1 Error while reading cell "tempsenseInst_error" (byte position 524262): Unknown layer/datatype in boundary, layer=14 type=0 Total of 10 warnings. Total of 10 warnings. Total of 10 warnings. Traceback (most recent call last): File "/OpenFASOC/openfasoc/generators/temp-sense-gen/tools/temp-sense-gen.py", line 223, in shutil.copyfile( File "/usr/lib/python3.10/shutil.py", line 254, in copyfile with open(src, 'rb') as fsrc: FileNotFoundError: [Errno 2] No such file or directory: 'tools/../flow/reports/sky130hd/tempsense/6_final_lvs.rpt' make: *** [Makefile:48: sky130hd_temp] Error 1 [test_loop.txt](https://github.com/idea-fasoc/OpenFASOC/files/13312681/test_loop.txt)
> 
> Please check the log and terminal output

@ripudamank2 Looks like you magic command failed because of permissions or issues with the magic tool (please refer to the bold and italic text that I highlighted above) thus not generating the lvs and drc reports. 
Are you able to work with magic in standalone mode outside OpenFASoC?

@hareyakana I don't think that is the case, the shutil library is trying to copy the file `tools/../flow/reports/sky130hd/tempsense/6_final_lvs.rpt` to the `work` directory but it couldn't find it thus throwing an error. Please refer to my comment above

ripudamank2, 2023-11-11:
yes i can open magic with and without root user. But there is no such file called sed5azpYH in magic directory. 
[image]

saicharan0112, 2023-11-12:
No I don't think that's a pre-defined file. Magic is trying to create some temporary file in this location and since this directory owner is `root` you don't have permissions to do that.

ripudamank2, 2023-11-12:
installed pdk locally and still phasing the same problem

ripudamank2, 2023-11-13:
Updated 0 paths from the index
CIF file read warning: CIF style sky130(): units rescaled by factor of 5 / 1
Error while reading cell "tempsenseInst_error" (byte position 523364): Unknown layer/datatype in boundary, layer=14 type=0
CIF file read warning: CIF style sky130(): units rescaled by factor of 5 / 1
Error while reading cell "tempsenseInst_error" (byte position 523364): Unknown layer/datatype in boundary, layer=14 type=0
Warning:  Ports "r_VIN(2)" and "r_VIN(1)" are electrically shorted.
Warning:  Ports "r_VIN(2)" and "r_VIN(0)" are electrically shorted.
Total of 10 warnings.
Warning:  Ports "r_VIN(2)" and "r_VIN(1)" are electrically shorted.
Warning:  Ports "r_VIN(2)" and "r_VIN(0)" are electrically shorted.
Warning:  Ports "r_VIN(2)" and "r_VIN(1)" are electrically shorted.
Warning:  Ports "r_VIN(2)" and "r_VIN(0)" are electrically shorted.
Total of 10 warnings.
Warning:  Ports "r_VIN(2)" and "r_VIN(1)" are electrically shorted.
Warning:  Ports "r_VIN(2)" and "r_VIN(0)" are electrically shorted.
Warning:  Ports "r_VIN(2)" and "r_VIN(1)" are electrically shorted.
Warning:  Ports "r_VIN(2)" and "r_VIN(0)" are electrically shorted.
Total of 10 warnings.
Warning:  Ports "r_VIN(2)" and "r_VIN(1)" are electrically shorted.
Warning:  Ports "r_VIN(2)" and "r_VIN(0)" are electrically shorted.
Traceback (most recent call last):
  File "/home/embedded/Documents/OpenFASOC/openfasoc/generators/temp-sense-gen/tools/temp-sense-gen.py", line 235, in <module>
    shutil.copyfile(
  File "/usr/lib/python3.10/shutil.py", line 254, in copyfile
    with open(src, 'rb') as fsrc:
FileNotFoundError: [Errno 2] No such file or directory: 'tools/../flow/reports/sky130hd/tempsense/6_final_lvs.rpt'
make: *** [Makefile:48: sky130hd_temp] Error 1

[temp_log.txt](https://github.com/idea-fasoc/OpenFASOC/files/13341211/temp_log.txt)

please check the log . 

This happened after installing pdk locally.

closed 2024-10-01 by chetanyagoyal