idea-fasoc/OpenFASOC #259: google colab ubuntu 22 LDO notebook

alibillalhammoud, 2023-10-28:
Since google colab transitioned to ubuntu 22, the setup code no longer works. Below is the correct setup code so that the notebook runs as expected. 
```
# install all tools and dependencies
import os
import pathlib
import sys
!apt install -y ngspice time
!curl -Ls https://micro.mamba.pm/api/micromamba/linux-64/latest | tar -xvj bin/micromamba
conda_prefix_path = pathlib.Path('conda-env')
CONDA_PREFIX = str(conda_prefix_path.resolve())
%env CONDA_PREFIX={CONDA_PREFIX}
!bin/micromamba create --yes --prefix $CONDA_PREFIX
!bin/micromamba install --yes --prefix $CONDA_PREFIX \
                        --channel litex-hub \
                        --channel main \
                        open_pdks.sky130a \
                        openroad \
                        netgen \
                        magic \
                        yosys \
                        klayout
!python -m pip install matplotlib ltspice pyyaml click gdstk cairosvg svgutils pillow mako
# clone OpenFASOC repo and setup env
!git clone https://github.com/alibillalhammoud/OpenFASOC
OPENFASOC_ROOT=str(pathlib.Path('OpenFASOC').resolve())
LDO_ROOT=OPENFASOC_ROOT+"/openfasoc/generators/ldo-gen/"
!cp OpenFASOC/openfasoc/generators/ldo-gen/blocks/sky130hvl/gds/capacitor_test_nf.gds OpenFASOC/docs/source/notebooks/ldo-gen
!cp OpenFASOC/openfasoc/generators/ldo-gen/blocks/sky130hvl/gds/PMOS.gds OpenFASOC/docs/source/notebooks/ldo-gen
!cp OpenFASOC/openfasoc/generators/ldo-gen/blocks/sky130hvl/gds/PT_UNIT_CELL.gds OpenFASOC/docs/source/notebooks/ldo-gen
!cp OpenFASOC/openfasoc/generators/ldo-gen/blocks/sky130hvl/gds/vref_gen_nmos_with_trim.gds OpenFASOC/docs/source/notebooks/ldo-gen
!cp OpenFASOC/openfasoc/generators/ldo-gen/blocks/sky130hvl/gds/LDO_COMPARATOR_LATCH.gds OpenFASOC/docs/source/notebooks/ldo-gen
!cp OpenFASOC/docs/source/notebooks/aux_files/def2gds.py OpenFASOC/docs/source/notebooks/ldo-gen
!cp OpenFASOC/docs/source/notebooks/aux_files/dbtodef.tcl OpenFASOC/docs/source/notebooks/ldo-gen
!cp OpenFASOC/openfasoc/common/platforms/sky130hvl/gds/sky130_fd_sc_hvl.gds OpenFASOC/docs/source/notebooks/ldo-gen
!cp OpenFASOC/openfasoc/common/platforms/sky130hvl/fill.json OpenFASOC/docs/source/notebooks/ldo-gen
%env PDK_ROOT=/content/conda-env/share/pdk
os.environ['PATH'] += ":/content/conda-env/bin"
```

saicharan0112, 2023-11-03:
Thanks @alibillalhammoud , can you send a PR with these updates please?

chetanyagoyal, 2024-06-13:
Can this be closed?