idea-fasoc/OpenFASOC #264: ldo generator fails at detail_place.tcl file

saicharan0112, 2023-11-07:
https://github.com/idea-fasoc/OpenFASOC/actions/runs/6753021994/job/18359047461#step:4:1983

msaligane, 2023-11-10:
It looks like the comparator is not being placed: 
```
Components of this program may be licensed under more restrictive licenses which must be honored.
[ERROR DPL-0017] cannot place instance cmp1
```

chetanyagoyal, 2023-12-06:
@harshkhandeparkar it seems like the call to the `detailed_placement` function (after `place_pt_unit` ) is where the error occurs

harshkhandeparkar, 2023-12-20:
This is what I have found so far:
1. The `detail_place.tcl` script first calls the `custom_place.tcl` script that is supposedly places the `cmp1` instance on [this line](https://github.com/idea-fasoc/OpenFASOC/blob/e7f3d20f9992c62b8976c5accf813a87a7b01b7f/openfasoc/generators/ldo-gen/flow/scripts/openfasoc/custom_place.tcl#L7).
2. After the custom placement, the OpenROAD command [`detailed_placement`](https://openroad.readthedocs.io/en/latest/main/src/dpl/README.html) is called.
3. If custom placement is disabled, the `detailed_placement` command simply fails to place `cmp1` with the error `[ERROR DPL-0017] cannot place instance cmp1.` (presumably by trying to place the already placed `cmp1`)
4. If custom placement is enabled but `detailed_placement` is disabled, the flow throws the following error: `Cell cmp1 with height 8140 is taller than any row.` (Presumably because `cmp1` is placed but violates some rules)
5. The command linked in point 1 sets the status of the cell to `PLACED` using the `-status` flag of the `place_cell` command. If the status is instead set to `FIRM`, the error mentioned in point 4 occurs even with `detailed_placement` enabled. (Presumably because setting the status to `FIRM` doesn't cause `cmp1` to be placed again but still violates the rules)

harshkhandeparkar, 2023-12-20:
Perhaps the size of each row was increased in OpenROAD?

harshkhandeparkar, 2023-12-20:
Also the GDS of the `cmp1` instance which is a custom block (`LDO_COMPARATOR_LATCH`) shows that the cell is of size ~19 micrometer x ~8 micrometer. Is this correct?

harshkhandeparkar, 2023-12-22:
I now understand that the cell is taller than a row, but the error is not supposed to be shown since it is custom-placed. I will investigate further on how to remove the error.

harshkhandeparkar, 2024-01-14:
While this issue has been fixed in #288, the DRC and LVS are [failing](https://github.com/idea-fasoc/OpenFASOC/actions/runs/7518454277/job/20465821168?pr=288).

harshkhandeparkar, 2024-01-18:
I found the reason for LVS failing:
```
Warning:  Ports "VSS" and "VDD" are electrically shorted.
```
Investigating further to find where the short occurs.

PR #288 (2024-08-22): Fixed LDO Flow
Closes #264 

### Changes
- [x] Changed the `LDO_COMPARATOR_LATCH` site to the double-height site `unithvdbl` provided by sky130hvl.
- [x] Changed the `PT_UNIT_CELL` site to `unithv` from `unithd`, which is meant for sky130hd.
- [x] Increased `VREG` voltage domain size by 1 micron on the x-axis on both sides. This prevents the core area rows from reaching the bottom (See Fig. 1 and Fig. 2)
- [x] Fixed the `LDO_COMPARATOR_LATCH` placement position, which was wrong because of the `MY` (mirror along y-axis) orientation.
- [x] Fixed the `save_images.tcl` script. The upstream ORFS ch [... cut]
closed 2024-08-22 by msaligane