idea-fasoc/OpenFASOC #287: Install NGSpice From Conda

harshkhandeparkar, 2023-12-24:
LiTeX-Hub has [ngspice](https://anaconda.org/litex-hub/ngspice) and [klayout](https://anaconda.org/litex-hub/klayout) that seem to be up to date.

The [dependencies.sh](https://github.com/idea-fasoc/OpenFASOC/blob/main/dependencies.sh) script installs these from the source. The installation would be greatly sped up if these packages were also installed using conda. Perhaps this could be kept as an option in case the latest versions are required for testing. By default, the conda versions can be installed to make local environment setup faster.

chetanyagoyal, 2024-06-09:
I think the installation time is offset by the customizability of installation from source, which might be desirable for developers working in local envs

curtisma, 2024-06-09:
We started work on adding klayout to [conda-forge](https://conda-forge.org/).  It is really close to being complete.

Check out the following PR with the klayout Conda recipe in-progress:
https://github.com/conda-forge/staged-recipes/pull/22419

curtisma, 2024-06-09:
ngspice is already available from the [conda-forge](https://conda-forge.org/packages/) channel.

[image]