idea-fasoc/OpenFASOC #294: Can't run docs/source/notebooks/glayout/GLayout_Cells.ipynb

d-m-bailey, 2024-03-08:
1. The `Run in Colab` link points to a [different file](https://colab.research.google.com/github/idea-fasoc/OpenFASOC/blob/main/docs/source/notebooks/temp-sense-gen/temp_sense_genCollab.ipynb). 
2. `glayout` is not loaded as a module, which causes an error in `1.3. Importing Libraries and Utility Functions`
```
ModuleNotFoundError                       Traceback (most recent call last)
[<ipython-input-31-fc1af012fbf9>](https://localhost:8080/#) in <cell line: 1>()
----> 1 from glayout.pdk.sky130_mapped import sky130_mapped_pdk as sky130
      2 from glayout.pdk.gf180_mapped  import gf180_mapped_pdk  as gf180
      3 import gdstk
      4 import svgutils.transform as sg
      5 import IPython.display

ModuleNotFoundError: No module named 'glayout.pdk.sky130_mapped'
```
3. Creating link to glayout in step 1.1 as below, loads the module.
```
!ln -s OpenFASOC/openfasoc/generators/gdsfactory-gen/glayout .
```
4. This results in a different error `sky130_mapped_pdk.cell_decorator_settings.cache=False` where `cell_decorator_settings` does not have the `cache` object.
```
ValueError                                Traceback (most recent call last)
[<ipython-input-34-fc1af012fbf9>](https://localhost:8080/#) in <cell line: 1>()
----> 1 from glayout.pdk.sky130_mapped import sky130_mapped_pdk as sky130
      2 from glayout.pdk.gf180_mapped  import gf180_mapped_pdk  as gf180
      3 import gdstk
      4 import svgutils.transform as sg
      5 import IPython.display

2 frames
[/content/glayout/pdk/sky130_mapped/__init__.py](https://localhost:8080/#) in <module>
      3 """
      4 
----> 5 from glayout.pdk.sky130_mapped.sky130_mapped import sky130_mapped_pdk

[/content/glayout/pdk/sky130_mapped/sky130_mapped.py](https://localhost:8080/#) in <module>
     50 # set the grid size
     51 sky130_mapped_pdk.gds_write_settings.precision = 5*10**-9
---> 52 sky130_mapped_pdk.cell_decorator_settings.cache=False
     53 sky130_mapped_pdk.gds_write_settings.flatten_invalid_refs=False

[/usr/local/lib/python3.10/dist-packages/pydantic/main.py](https://localhost:8080/#) in __setattr__(self, name, value)
    793         elif self.model_config.get('extra') != 'allow' and name not in self.model_fields:
    794             # TODO - matching error
--> 795             raise ValueError(f'"{self.__class__.__name__}" object has no field "{name}"')
    796         elif self.model_config.get('extra') == 'allow' and name not in self.model_fields:
    797             if self.model_extra and name in self.model_extra:

ValueError: "CellDecoratorSettings" object has no field "cache"
```
This may be due to the removal of the assignment in `openfasoc/generators/gdsfactory-gen/glayout/pdk/mappedpdk.py`
```
    def from_gf_pdk(
        cls,
        gfpdk: Pdk,
        **kwargs
    ):
...
#        passargs["cell_decorator_settings"]=gfpdk.cell_decorator_settings
```

Removing the line causing the error, results in other errors. Maybe there's an older version that works?
```
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
[<ipython-input-44-fc1af012fbf9>](https://localhost:8080/#) in <cell line: 1>()
----> 1 from glayout.pdk.sky130_mapped import sky130_mapped_pdk as sky130
      2 from glayout.pdk.gf180_mapped  import gf180_mapped_pdk  as gf180
      3 import gdstk
      4 import svgutils.transform as sg
      5 import IPython.display

2 frames
[/content/glayout/pdk/sky130_mapped/__init__.py](https://localhost:8080/#) in <module>
      3 """
      4 
----> 5 from glayout.pdk.sky130_mapped.sky130_mapped import sky130_mapped_pdk

[/content/glayout/pdk/sky130_mapped/sky130_mapped.py](https://localhost:8080/#) in <module>
     51 sky130_mapped_pdk.gds_write_settings.precision = 5*10**-9
     52 #sky130_mapped_pdk.cell_decorator_settings.cache=False
---> 53 sky130_mapped_pdk.gds_write_settings.flatten_invalid_refs=False

[/usr/local/lib/python3.10/dist-packages/pydantic/main.py](https://localhost:8080/#) in __setattr__(self, name, value)
    793         elif self.model_config.get('extra') != 'allow' and name not in self.model_fields:
    794             # TODO - matching error
--> 795             raise ValueError(f'"{self.__class__.__name__}" object has no field "{name}"')
    796         elif self.model_config.get('extra') == 'allow' and name not in self.model_fields:
    797             if self.model_extra and name in self.model_extra:

ValueError: "GdsWriteSettings" object has no field "flatten_invalid_refs"
```

d-m-bailey, 2024-03-09:
Ok. So during the first installation, there was this message
```
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
gf180 0.1.0 requires gdsfactory<7.17,>=7.16.0, but you have gdsfactory 7.7.0 which is incompatible.
sky130 0.10.1 requires gdsfactory==7.16.0, but you have gdsfactory 7.7.0 which is incompatible.
```
So I bumped the gdsfactory version to 7.16.0 which seems to have broken everything.

harshkhandeparkar, 2024-03-09:
> glayout is not loaded as a module, which causes an error in

This is probably because the installation steps seem to be missing a `cd` step. I'll make PR fixing this soon.

Until then, can you try using the installation steps in the first cell of [this](https://github.com/idea-fasoc/OpenFASOC/blob/main/docs/source/notebooks/glayout/glayout_opamp.ipynb) notebook?

cc: @andylithia

closed 2024-03-14 by msaligane