idea-fasoc/OpenFASOC #320: glayout upgrade to gdsfactory 8?

curtisma, 2024-06-09:
gdsfactory 8 was released today with major updates moving the backend from gdstk to klayout.

Should we update the glayout [dependency from gdsfactory v7.7.0](https://github.com/idea-fasoc/OpenFASOC/blob/7dc5eb42cec94c02b74e72483df6fdc2b2603fb9/openfasoc/generators/glayout/setup.py#L21C2-L21C29) to v8?

It would be a good idea to update it to this latest major revisions before most people start working on glayout generators for the Chipathon.

[image]

chetanyagoyal, 2024-06-13:
This is something we want to look into. Currently, due to the significant leap in versions, there are some broken imports and inconsistencies. Can we have some guidance on where things are now 

To be more specific `gdsfactory.functions` had `transform` and `move` as importable functions (which were wrappers for more involved functions) but these don't exist anymore. There are general issues like this for which help is required @curtisma

msaligane, 2024-06-24:
How much stuff in our repo would make sense for us to move to gdsfactory ? wouldn't it make sense to just use Klayout directly?
cc @joamatab @tvt173

tvt173, 2024-06-25:
hi @msaligane , sure i'd be happy to take a look or help work through this with you or someone from your team later this week

> wouldn't it make sense to just use Klayout directly?

yes, of course you _could_, but to me the question is similar to "wouldn't it make more sense to just write this all in c or machine code?" 🙂 . the question isn't whether or not it's _possible_ but how it affects your _productivity_

gdsfactory is meant to provide a more convenient interface to create designs using a code-driven flow. klayout _does_ provide a python interface, but i find it a pain to use. i think that what we've built on top of it makes it much simpler. so i suppose the question hinges on whether or not you agree and find value in the accessibility and productivity boost that gdsfactory brings

chetanyagoyal, 2024-06-28:
@tvt173 we tried porting to gdsfactory 8, and due to all the syntactic changes that need to be done due to backend change, a lot of extra work is required. However, generally, migrating to the new version (or klayout entirely) is something that should be done due to the additional functionality. We might be pursued a bit further into things. Do you agree @msaligane ?