idea-fasoc/OpenFASOC #360: cannot import name 'diff_pair' from 'glayout.flow.blocks'

circuitmuggle, 2024-12-28:
I tried to run the glayout_opamp,ipynb in both google runtime and my local runtime.

https://colab.research.google.com/github/idea-fasoc/OpenFASOC/blob/main/docs/source/notebooks/glayout/glayout_opamp.ipynb#scrollTo=ozTFXBekORtd

The first two code blocks executed correctly. When I moved to the 3rd code block. (Interactive Primitive Generation in SKY130
The following cell demonstrates the different PCell and Utility generators on the Sky130 PDK.)

both runtime (google and local) report the below errors while I checked the corresponding folders are there under FASOC/openfasoc/generators/glayout/glayout/flow/blocks/elementary/diff_pair. Not sure why diff_pair cannot be imported. 

[image]
[image]

circuitmuggle, 2024-12-28:
replace glayout package path from "pip install glayout" path solved the above issue, pending fix.
[image]

msaligane, 2024-12-29:
Thanks!
@alibillalhammoud and @harshkhandeparkar can you please review?

circuitmuggle, 2024-12-30:
I looked at the source code and realized that maybe folder composite and elementary are for future release? Close for now as I am not sure what's the intention for next release. Easiest fix will be just add elementary back to glayout_opamp.ipynb so that it becomes

from glayout.flow.blocks.elementary import diff_pair

it will align with the rest of this git repository.

closed 2024-12-30 by circuitmuggle

circuitmuggle, 2024-12-30:
re-submit pull request to fix this issue https://github.com/idea-fasoc/OpenFASOC/pull/362